thelastoutpostworkshop/arduino-maker-workshop #3: Unable to select any board from drop down options

Jetlec, 2024-12-22:
Hello, I am unable to functionally select any board from the drop down options. I do get the options but not the ability to enable my selection. I have a working Waveshare ESP32S3 128 plugged in that works with the VSCODE/ESPRESSIF IDF with no issues there.
[image]

thelastoutpostworkshop, 2024-12-22:
There is a folder created .vscode and inside it there is an arduino.json file.
Copy the content here also a screen shot of the board manager like the picture below.
Also the arduino Extension output content.
[image]

thelastoutpostworkshop, 2025-01-03:
Closing stale issue

closed 2025-01-03 by thelastoutpostworkshop