thelastoutpostworkshop/arduino-maker-workshop #4: Does not work in Linux

wmerkens, 2024-12-22:
Install the Extension in Windows, and it works, but in Linux, it does not.

This error is generated

`[error] Unable to resolve nonexistent file 'vscode-userdata:/home/willm/.config/Code/User/globalStorage/thelastoutpostworkshop.arduino-maker-workshop': Error: Unable to resolve nonexistent file 'vscode-userdata:/home/willm/.config/Code/User/globalStorage/thelastoutpostworkshop.arduino-maker-workshop'
    at _ot.resolve (vscode-file://vscode-app/usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js:3108:60549)
    at async BTt (vscode-file://vscode-app/usr/share/code/resources/app/out/vs/workbench/workbench.desktop.main.js:480:44682)`

If I go to /home/willm/.config/Code/User/globalStorage  the folder thelastoutpostworkshop.arduino-maker-workshop does not exist.

Also in /home/willm/.vscode/extensions/thelastoutpostworkshop.arduino-maker-workshop-0.4.4/arduino_cli

you only show win32 installed

thelastoutpostworkshop, 2024-12-22:
It's not supported on Linux yet, I need help for that : https://github.com/thelastoutpostworkshop/arduino-maker-workshop/discussions/2

davidthefuturist, 2024-12-23:
I also don't seem to be able to get it working on macOS. Is that also not supported yet?

thelastoutpostworkshop, 2024-12-23:
No it's not supported yet

thelastoutpostworkshop, 2024-12-30:
Linux-x64 is now supported on version 0.5.0