thelastoutpostworkshop/arduino-maker-workshop #6: Unable to create a new sketch

kevin-hart, 2024-12-23:
unable to create new sketch -  see screen capture below.

[image]

extension is reporting following errors.

[image]

and when trying to add 

Boards Manager is showing that I have the following boards installed.

[image]

however, when I try to select a board I get the following errors.  

[image]

I created a new Profile and installed the Arduino Maker Workshop extension - which then installed its dependencies 

[image]

thelastoutpostworkshop, 2024-12-23:
Are you on Windows ?
If yes, copy what's in the "Arduino Extension" output Window 
[image]
 and also what's in the "Arduino CLI" output Window.
[image]

kevin-hart, 2024-12-23:
> Are you on Windows ? If yes, copy what's in the "Arduino Extension" output Window [...]

Yes - I'm running windows (Windows 10 22H2 to be specific)   -  See below for VS code version

[image]

Outputs requested (I think)

Arduino Extension started
Arduino CLI is ready, path: c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
Arduino Config file is good
Arduino Web view ready
Set Board FQBN=:arduino:avr:uno
Set Board FQBN=:arduino:avr:uno
Set Board FQBN=:arduino:avr:uno

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
version --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
core update-index

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
config dump --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
core update-index

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
lib search --omit-releases-details --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
lib list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
outdated --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board search --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
core search -a --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
config dump --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\kevin\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.6-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

thelastoutpostworkshop, 2024-12-23:
I think I know what's happening, open Visual Studio Code in a folder where you want your sketch to be stored, for example in the your documents folder and try creating a new sketch.

kevin-hart, 2024-12-23:
Thanks that works.   Will have a play.Kevin Sent from Outlook for AndroidFrom: thelastoutpostworkshop ***@***.***>Sent: Monday, December 23, 2024 6:43:50 pmTo: thelastoutpostworkshop/arduino-maker-workshop ***@***.***>Cc: Kevin Hart ***@***.***>; Author ***@***.***>Subject: Re: [thelastoutpostworkshop/arduino-maker-workshop] Unable to create a new sketch (Issue #6)
I think I know what's happening, open Visual Studio Code in a folder where you want your sketch to be stored, for example in the your documents folder and try creating a new sketch.

peno64, 2024-12-28:
I have exactly the same problem. However starting in another folder where I can write in does not help.
I also have the ENOENT: no such gile or directory, open ''

And I can't create a new sketch because the option is just not there. It is also on the exact same windows version.

[image]
[image]

[image]

peno64, 2024-12-28:
[image]

thelastoutpostworkshop, 2024-12-28:
If you open vscode in a folder where you have write permission and you get the error, post the content of the output windows "Arduino CLI" and "Arduino Extension" here

peno64, 2024-12-28:
Arduino CLI:

Running Arduino CLI...
c:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\arduino-cli.exe
version --json

Running Arduino CLI...
c:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\arduino-cli.exe
core update-index

Running Arduino CLI...
c:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\arduino-cli.exe
config dump --json

Running Arduino CLI...
c:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\arduino-cli.exe
core update-index

Running Arduino CLI...
c:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Running Arduino CLI...
c:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\arduino-cli.exe
outdated --json

Arduino Extension:

Arduino Extension started
Arduino CLI is ready, path: c:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\arduino-cli.exe
Arduino Config file is good
Arduino Web view ready
New Port=:COM4

peno64, 2024-12-28:
I now installed it on a fresh new windows 11 computer and exactly the same problem. Also note when I go to "Board Selection" that no boards are shown at all. If I go to board configuration I have there the button select a board first but again nothing happens

peno64, 2024-12-28:
Boards are installed...

[image]

thelastoutpostworkshop, 2024-12-28:
Open a command shell where the arduino-cli is located : 
`c:\Users\mail.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\arduino-cli.exe`
and run this command : `arduino-cli sketch new MySketch`
paste the output of the command

peno64, 2024-12-28:
Seems to work fine: 

C:\Windows\System32>cd "C:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32"

C:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32>arduino-cli sketch new mysketch
Sketch created in: C:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\mysketch

C:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32>cd mysketch

C:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\mysketch>dir
 Volume in drive C is Windows
 Volume Serial Number is 243A-4A88

 Directory of C:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\mysketch

28/12/2024  17:04    <DIR>          .
28/12/2024  17:04    <DIR>          ..
28/12/2024  17:04                35 mysketch.ino
               1 File(s)             35 bytes
               2 Dir(s)  645.877.010.432 bytes free

C:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.7-win32-x64\arduino_cli\win32\mysketch>

thelastoutpostworkshop, 2024-12-28:
if you open that sketch in vscode is the extension working ?

peno64, 2024-12-28:
I opened it via File, Open and I get:

[image]

But the workshop is still an empty screen:

[image]

peno64, 2024-12-28:
When I hoover over Compile at says that it is disabled, not an Arduino Project:

[image]

peno64, 2024-12-28:
When I do Help I see that Visual Studio Code is a version from last week. Would it be that the extension doesn't work anymore with the latest VS version?

[image]

thelastoutpostworkshop, 2024-12-28:
It's compatible with the latest version of VSCode.  In the folder of that sketch, there should a .vscode folder and inside it an Arduino.json file, if it's there, paste the content here

peno64, 2024-12-28:
No, there is no subfolder .vscode
even not hidden

thelastoutpostworkshop, 2024-12-28:
Here is a test version of the extension that log more errors : https://drive.google.com/file/d/1EKUWHStfTiwXvidHHeSr0pkt2maRgmyH/view?usp=drive_link
It's a .vsix file in a zip.
- uncompress the zip
- uninstall the official extension in vscode
- install this test version in vscode with the command "Extension: Install from VSIX... (CTRL+SHIFT+P)
- reproduce the error and paste the content of the "Arduino Extension" output

peno64, 2024-12-28:
send a request to be able to download the file

peno64, 2024-12-28:
Arduino Extension started on win32
Problem : cannot get projectFullPath
Arduino CLI is ready, path: c:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.93\arduino_cli\win32\arduino-cli.exe
Arduino Config file is good
Arduino Web view ready

thelastoutpostworkshop, 2024-12-28:
So the errors cannot get projectFullPath occurs when the extension attempt to get the path of your workspace folder in vscode : https://code.visualstudio.com/docs/editor/workspaces
This is strange that your vscode has no workspace folder, it's not supposed to be.

peno64, 2024-12-28:
Ok, I did the following from your link: "File > Open Folder... menu, and then selecting a folder."

and now I have the create a new sketch input:

[image]

So after you freshly installed visual code it seems there is no workspace defined yet...

peno64, 2024-12-28:
However... The button "SELECT A BOARD FIRST" is still doing nothing and nothing extra in the Arduino Extension output...

peno64, 2024-12-28:
Can you also add debugging when "SELECT a BOARD FIRST" is chosen to see why nothing happens?

peno64, 2024-12-28:
When I click on "SELECT A BOARD FIRST", the following screen first appears but after a second it becomes empty:

[image]

And there are boards installed:

[image]

thelastoutpostworkshop, 2024-12-28:
You should now have an arduino.json file inside the .vscode folder, paste the content

peno64, 2024-12-28:
{
  "port": "",
  "configuration": "",
  "output": "build",
  "board": "",
  "programmer": "",
  "useProgrammer": false,
  "configurationRequired": false
}

thelastoutpostworkshop, 2024-12-29:
Try running this commands where the arduino cli is located : 
`arduino-cli board search > boardsearch.json`
This will create the file boardsearch.json, zip it and paste it here.

peno64, 2024-12-29:
[boardsearch.json](https://github.com/user-attachments/files/18268908/boardsearch.json)

thelastoutpostworkshop, 2024-12-29:
Sorry, i forgot an option in the command, it is:
`arduino-cli board search --json > boardsearch.json`
Also run this command paste the output here:
`arduino-cli config dump`

peno64, 2024-12-29:
[boardsearch.zip](https://github.com/user-attachments/files/18269632/boardsearch.zip)

arduino-cli config dump:

board_manager:
    additional_urls: []
directories:
    data: C:\Users\mail\AppData\Local\Arduino15
    downloads: C:\Users\mail\AppData\Local\Arduino15\staging
    user: C:\Users\mail\Documents\Arduino

thelastoutpostworkshop, 2024-12-29:
I have create another vsix : https://drive.google.com/file/d/1vtWPBzSvDI3lpVfEmglQ2zEDbZ0nGar9/view?usp=drive_link
Reproduce the error and paste the output of "Arduino Extension"

peno64, 2024-12-29:
Arduino Extension started on win32
Workspace is c:\brol\arduino\test
Arduino CLI is ready, path: c:\Users\mail\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.4.94\arduino_cli\win32\arduino-cli.exe
Arduino Config file is good
Arduino Web view ready
Sending Board search...
Board search sent

I am afraid that this won't give much information. The last two lines appear when I chose to select the board.

thelastoutpostworkshop, 2024-12-29:
I found what the problem is. I am solving it and will send you another .vsix

thelastoutpostworkshop, 2024-12-29:
here is the new vsix to test : https://drive.google.com/file/d/1pHM0tUC1KuKBAFVu_crGk79QscpigA_o/view?usp=drive_link

peno64, 2024-12-29:
Yes! That solves the issue.
Thank you very much. I hope this fix will also help others.
And maybe also add a test in your code if no workspace is selected that an apropriate message is given.

thelastoutpostworkshop, 2024-12-29:
Thanks for having the patience to help make the extension better.
Yes I have added a message when no workspace is selected.
I will push the official version with the fix soon.

thelastoutpostworkshop, 2024-12-30:
Extension v0.5.0 pushed on the vscode marketplace that solved this problem.

closed 2024-12-30 by thelastoutpostworkshop