thelastoutpostworkshop/arduino-maker-workshop #7: CLI: Failed to get library available

mspcbs, 2024-12-24:
Library Manager , CLI: Failed to get library available

thelastoutpostworkshop, 2024-12-24:
What is the OS of your computer ?
Copy the output of the "Arduino CLI" output window 
[image]
and the "Arduino Extension" Window 
[image]

Maxbeth2, 2024-12-29:
I get the same problem, here is the output from the Arduino CLI:

```
Running Arduino CLI...
/Users/max/.vscode/extensions/thelastoutpostworkshop.arduino-maker-workshop-0.4.92-darwin-arm64/arduino_cli/darwin/arm64/arduino-cli
version --json
Failed to run command: spawn /Users/max/.vscode/extensions/thelastoutpostworkshop.arduino-maker-workshop-0.4.92-darwin-arm64/arduino_cli/darwin/arm64/arduino-cli EACCES
Command failed with code -13.
```

And the extension:

```
Arduino Extension started on darwin
Cannot get CLI version
```

thelastoutpostworkshop, 2024-12-29:
I have created a new issue since your on a Mac : https://github.com/thelastoutpostworkshop/arduino-maker-workshop/issues/12

closed 2024-12-29 by thelastoutpostworkshop