thelastoutpostworkshop/arduino-maker-workshop #8: intelli sense does not work.

marvinschwartz, 2024-12-24:
After installation intelli sense does not recognise any Arduino commands
[image]

This include auto complete.

thelastoutpostworkshop, 2024-12-24:
Intellisense information is created after compile

marvinschwartz, 2024-12-24:
Not sure if you understand. Those little red squiggles are there before compile and Arduino commands are not auto-completed or recognised . 
Compile and Upload work fine. 
I do not have the Arduino extension from Microsoft installed as it is no longer available.

thelastoutpostworkshop, 2024-12-24:
yes I know, please attached the c_cpp_properties.json file that is stored in the .vscode folder in your project 
[image]

AJ-Smoothie, 2024-12-27:
Same issue here. I compared the c_cpp_properties.json file with the file from the deprecated Arduino extension, and found the problem.

There is one line missing

```
"forcedInclude": [
              "C:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino\\Arduino.h"
          ],
```

The include that this extension makes is this: 
```"C:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages/**",```

which is the same path. I'm guessing the /** implies a search of all files, but for some reason it is unable to grab Arduino.h.

thelastoutpostworkshop, 2024-12-27:
Currently intellisense will add what's included as header files in the code, I will look at adding defaults based on the platform selected.

AJ-Smoothie, 2024-12-27:
> After installation intelli sense does not recognise any Arduino commands

Quick fix! Just throw in #include <Arduino.h> at the top of your sketch!

closed 2025-01-03 by thelastoutpostworkshop

thelastoutpostworkshop, 2025-01-22:
The way you describe it, is like multi-root workspace in vscode : https://code.visualstudio.com/docs/editor/multi-root-workspaces

Am I correct ?

ZUIcat, 2026-01-07:
> > Currently intellisense will add what's included as header files in the code, I will look at adding defaults based on the platform selected.
> 
> Quick fix! Just throw in `#include <Arduino.h>` at the top of your sketch!

That's amazing—it solved my issue. Why is that? Does Arduino automatically add this header file by default, so my code compiles even without it, while VSCode doesn’t do that, causing IntelliSense to fail?

AJ-Smoothie, 2026-01-07:
> That's amazing—it solved my issue. Why is that? Does Arduino automatically add this header file by default, so my code compiles even without it, while VSCode doesn’t do that, causing IntelliSense to [...]

The Arduino IDE automatically includes this header during compilation. I also believe compiling through the CLI will include it as well, but I'm not 100% sure on the latter. Intellisense can't see this file if it's not explicitly included, hence the errors. Including it allows intellisense to see the file.