thelastoutpostworkshop/arduino-maker-workshop #10: Need option to setup custom sketchbook (library) directory

AJ-Smoothie, 2024-12-28:
If you have your sketchbook library in a custom location, this extension by default will not be able to find your libraries. **There is no option to add a custom path** either, that's why I'm opening this as an issue.

But I'm not going to leave you high and dry! I found a way to setup your own library directory. When you install this extension, it installs a version of the arduino-cli. There's a file in the that points to your sketchbook location (hence library location).

Navigate to `C:\Users\<username>\AppData\Local\Arduino15` and open `arduino-cli.yaml`, it should look something like this:

[image]

Replace the path after `user:` with your custom sketchbook location, and save the file. 

If your first compilation fails (because it can't find the libraries), it will not create a `c_cpp_properties.json` file under the `.vscode` folder of your sketch. In order to fix this, from the Arduino Maker Workshop menu, make sure you select `Compile(clean)`. This will generate that file, and also it will find your new directory. Now if you go to the library manager, your currently installed libraries should show up.

thelastoutpostworkshop, 2024-12-28:
Thanks for the tip, I have added it for enhancement

roboticboyer, 2025-01-11:
It's could be useful to have a menu where to set the library folder and the board/core folder such is done with the Portable version of the old IDE 1.8.x. 
Sometimes it's better to have multiple version of the cores or the libraries due to incompatibility of the revisions

yonatan8070, 2025-01-12:
Adding to this, I always create a new folder per-project and set the sketchbook location to it, so that I never have version conflicts between projects. I usually set this folder to be the sketch folder itself, so my `libraries` directory is next to the .ino files.

roboticboyer, 2025-01-17:
I think that a solution could be to add a menu to set the [build-profiles](https://arduino.github.io/arduino-cli/1.1/sketch-project-file/#build-profiles)
The purpose of build profiles is to control the project dependencies. When you use a build profile when compiling a sketch, the globally installed boards platforms and libraries are not used. Instead, a dedicated dependencies environment is use

thelastoutpostworkshop, 2025-01-17:
I will take a look at build profiles, thanks for the tips!

closed 2025-01-24 by thelastoutpostworkshop

roboticboyer, 2025-01-25:
Please provide an example or a short YT video how to use this functionality, which is very useful

AJ-Smoothie, 2025-01-26:
> Please provide an example or a short YT video how to use this functionality, which is very useful

Here ya go!

https://youtu.be/7lA5iIJi72c?si=9JeBfWxeZjZ2Lz_h&t=184