thelastoutpostworkshop/arduino-maker-workshop #11: Upload does not automatically compile

AJ-Smoothie, 2024-12-28:
If you click "Upload", it does not automatically compile a new file like we're used to. This is quite confusing because if you change some code and click "Upload", then it will upload the previously compiled file as if nothing ever happened, causing great confusion. You have to first manually click "Compile", then you can click "Upload" to upload your new code.

thelastoutpostworkshop, 2024-12-28:
I pushed version 0.4.8.
For now the build output will be invalidated when files are changed in the project or the board configuration is changed.
I have still have to implement the auto compile when upload is used on an invalidated build output.

AJ-Smoothie, 2025-01-03:
If you have shortcuts bound to Compile and Upload, this new update does not disable the button through the shorcuts.

thelastoutpostworkshop, 2025-01-03:
yes you are right

thelastoutpostworkshop, 2025-01-03:
Solved in v0.6.1 now available.
Upload will automatically recompile before uploading if needed

closed 2025-01-03 by thelastoutpostworkshop