thelastoutpostworkshop/arduino-maker-workshop #12: I get the same problem (on my m1 mac), here is the output from the Arduino CLI:

thelastoutpostworkshop, 2024-12-29:
I get the same problem (on my m1 mac), here is the output from the Arduino CLI:

```
Running Arduino CLI...
/Users/max/.vscode/extensions/thelastoutpostworkshop.arduino-maker-workshop-0.4.92-darwin-arm64/arduino_cli/darwin/arm64/arduino-cli
version --json
Failed to run command: spawn /Users/max/.vscode/extensions/thelastoutpostworkshop.arduino-maker-workshop-0.4.92-darwin-arm64/arduino_cli/darwin/arm64/arduino-cli EACCES
Command failed with code -13.
```

And the extension:

```
Arduino Extension started on darwin
Cannot get CLI version
```

_Originally posted by @Maxbeth2 in https://github.com/thelastoutpostworkshop/arduino-maker-workshop/issues/7#issuecomment-2564753471_

thelastoutpostworkshop, 2024-12-29:
The EACCES error means the arduino-cli don't have access to the folder.  
see this if that helps : https://stackoverflow.com/questions/38980338/eacces-permission-denied-in-vs-code-mac

rudeb0t, 2024-12-30:
@Maxbeth try the following from your command line:

```
chmod +x /Users/max/.vscode/extensions/thelastoutpostworkshop.arduino-maker-workshop-0.4.92-darwin-arm64/arduino_cli/darwin/arm64/arduino-cli
```

 You may need to change 0.4.92 to the current version of the extension that is installed on your Mac.

This is a known issue when publishing extensions from Windows. `vsce` on Windows strips out the executable bit from files that are supposed to have it. See: https://code.visualstudio.com/api/working-with-extensions/publishing-extension#why-does-vsce-not-preserve-file-attributes

Maxbeth2, 2024-12-30:
> @maxbeth try the following from your command line:
>
> ```
> chmod +x /Users/max/.vscode/extensions/thelastoutpostworkshop.arduino-maker-workshop-0.4.92-darwin-arm64/arduino_cli/darwin/arm64/arduino-cli [...]

This works, thanks both!

thelastoutpostworkshop, 2024-12-30:
@rudeb0t for Linux, I added a chmodSync to add execute permission to the arduino-CLI in getBundledArduinoCliPath()

rudeb0t, 2024-12-30:
I'll update my PR to do this for macOS as well.

closed 2025-01-10 by thelastoutpostworkshop