thelastoutpostworkshop/arduino-maker-workshop #15: Upload changes baud rate to 115200

peno64, 2024-12-30:
**Extension version**
Latest version of Arduino Maker Workshop

**Describe the bug**
Each time a sketch is uploaded to an arduino, the baud rate is changed to 115200 resulting also in that baud rate for the serial monitor. So if the sketch for example uses 9600 baud then after each upload the baud rate must be changed again in the serial monitor.

**Expected behavior**
The baud rate should be restored again to its original value after upload and before the monitor makes a new connection.

peno64, 2024-12-30:
I now also note that Line ending is also changed to None after each upload. I have it on CRLF and it should stay like that.

rudeb0t, 2024-12-30:
115200 is hard-coded. We will need to implement a panel for serial monitor settings.

peno64, 2024-12-30:
Is it not possible to read the value from the Baud rate input box before uploading and setting this value back after?

rudeb0t, 2024-12-30:
> Is it not possible to read the value from the Baud rate input box before uploading and setting this value back after?

The SerialMonitorApi does not expose anything that we can use for that unfortunately. So we have to set the values for port, baud rate, stop bits, etc when we create a new serial monitor tab. Also the baud rate for upload is based on the board config. It can be different from the baud rate of the serial monitor console.

thelastoutpostworkshop, 2025-01-03:
This new feature is now available in v0.6.1

thelastoutpostworkshop, 2025-01-09:
@rudeb0t do you think we can close this issue ?

rudeb0t, 2025-01-10:
Yes I think so.

closed 2025-01-10 by thelastoutpostworkshop