thelastoutpostworkshop/arduino-maker-workshop #18: Can't find include files

ringerattila, 2025-01-01:
The file in the same folder is ok, it seems for the compiler. But it can't find installed libraries, even if they are in the right place.
[image]

[image]
[image]

thelastoutpostworkshop, 2025-01-01:
Dis you compile your sketch ?  IntelliSense is generated after each compil

Joram3000, 2025-01-02:
I had the same issue 

[image]

as far as i can tell i added a bunch of libraries 

i seem to be able to fix it now with "../libraries/**" added to the settings thingy 
[image]

thelastoutpostworkshop, 2025-01-02:
very good, there will be an update to the extension soon, that will fix more intelliSense issues.

closed 2025-01-02 by thelastoutpostworkshop

Joram3000, 2025-01-10:
thank you ! awesome piece of software :) 
thank you for writing it!