thelastoutpostworkshop/arduino-maker-workshop #19: AMW vs ESP-IDF

dvukovic, 2025-01-07:
Hello, I discovered your extension today and installed it on my Win10 Dell Laptop.

I tried a simple test program and it seems to work.
I then tried ESP_Async_WebServer\examples\CaptivePortal to really test it out.
I kept getting two errors of #include files missing [#include <DNSServer.h>]

After a few hours I discovered I needed to uninstall ESP-IDF extension !

After I did that, every thing worked.

Thanks for listening.

thelastoutpostworkshop, 2025-01-07:
Thanks to have taken the time to solve it and reporting it.
I will test it on my side to see if there's any conflict between the two extensions.

thelastoutpostworkshop, 2025-01-10:
I installed the ESP-IDF extension and wasn't able to reproduce the errors you are having.
Everything worked fine on my side.

thelastoutpostworkshop, 2025-01-19:
Issue closed : stale

closed 2025-01-19 by thelastoutpostworkshop