thelastoutpostworkshop/arduino-maker-workshop #20: Incomplete/late serial output

andrew2221111, 2025-01-07:
Hello,

First of all, thank you very much for this extension!

I have an [Adafruit Feather RP2040 USB Host](https://learn.adafruit.com/adafruit-feather-rp2040-with-usb-type-a-host/overview) board, which is configured correctly through your extension. Everything works fine, compile, upload, etc.

Not sure if this is a bug or a setting related issue. However, looking at the serial monitor, I observed that the first few lines are cut out, not shown, when I (warm) reboot the board (from the dedicated button). Here is a comparison: this is what I see in ArduinoIDE:
[image]

And this is what I see in the extension:
[image]

My thinking is that something happens later then it should, either the reconnection taps in a bit later, or the capturing. Either way, I thought I open a ticket, in case it is worth investigating.

I have just installed the extension, so it's the latest version (0.6.1). Maybe worth mentioning, that I use VSCodium (1.96.2), not VSCode. If this causes my issue, then please feel free to close this.

Thank you.

thelastoutpostworkshop, 2025-01-07:
The connection to the serial monitor is not immediate and the code has the time to run before the connection is made.
I will see if anything can be done to speed that up.

andrew2221111, 2025-01-07:
Not sure if it would be too far fetched as an idea, or if it's even doableb, but here is one.

Since there is, a hardware communication (through the COM port), maybe that could be initialized first, before everything, start to capture the communication, then the rest of the UI elements and other functionality.

I'm aware that we are talking about microsecond units, maybe even less, and that might be influenced by one's hardware (e.g. PC specs), but in this case, matters. Thank you again.

andrew2221111, 2025-01-07:
Yes, which on reset/reboot, not sure how could be done, as the connection breaks completely due to re-initialization of hardware level stuff..

As a sidenote, I see in the serial monitor's repo that others reported speed related issues as well.. I guess not much we can do about it.

Anyway, appreciate it. Thank you.

thelastoutpostworkshop, 2025-01-10:
Closing this since there not much we can do about it

closed 2025-01-10 by thelastoutpostworkshop