thelastoutpostworkshop/arduino-maker-workshop #21: Intellisense Issue with SAMD Board (MKR WiFi 1010)

0x3b29, 2025-01-08:
--> Note that the extension is only compatible on Windows 64bits

**Extension version**
0.6.1

**Describe the bug**
When I start a fresh project, and select the Arduino MKR WiFi 1010 Board, and hit compile, the compilation is successfull but then Intellisense states that there are problems

**To Reproduce**
Steps to reproduce the behavior:
1. Start new project
2. Select board Arduino MKR WiFi 1010
3. Build
4. See Problems (Ctrl + Shift + M)

**Expected behavior**
No "Problems"

**Screenshots**
[image]

**Outputs**
`Running Arduino CLI...
c:\Users\oli\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.6.1-win32-x64\arduino_cli\win32\arduino-cli.exe
compile -v --no-color --fqbn arduino:samd:mkrwifi1010 --build-path c:\Users\oli\git\ArduinoMakerWorkshopMinimal\build c:\Users\oli\git\ArduinoMakerWorkshopMinimal --clean`

**Additional context**
The compilation is successfull, Ill add the compilation log as the first comment to not overload this description

0x3b29, 2025-01-08:
Compile project starting...

FQBN: arduino:samd:mkrwifi1010
Verwende das Board 'mkrwifi1010' von der Plattform im Ordner: C:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14
Verwendung des Kerns 'arduino' von Platform im Ordner: C:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14

Verwendete Bibliotheken erkennen ...
C:\Users\oli\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_MKRWIFI1010 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8054 -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino MKR WiFi 1010" -DUSE_BQ24195L_PMIC -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/ -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/ -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14\cores\arduino/api/deprecated -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14\cores\arduino/api/deprecated-avr-comp -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14\cores\arduino -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14\variants\mkrwifi1010 C:\Users\oli\git\ArduinoMakerWorkshopMinimal\build\sketch\ArduinoMakerWorkshopMinimal.ino.cpp -o nul
Funktionsprototypen werden generiert ...
C:\Users\oli\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_MKRWIFI1010 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8054 -DUSBCON -DUSB_MANUFACTURER="Arduino LLC" -DUSB_PRODUCT="Arduino MKR WiFi 1010" -DUSE_BQ24195L_PMIC -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/ -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/ -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14\cores\arduino/api/deprecated -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14\cores\arduino/api/deprecated-avr-comp -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14\cores\arduino -IC:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14\variants\mkrwifi1010 C:\Users\oli\git\ArduinoMakerWorkshopMinimal\build\sketch\ArduinoMakerWorkshopMinimal.ino.cpp -o C:\Users\oli\AppData\Local\Temp\2163899450\sketch_merged.cpp
C:\Users\oli\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives C:\Users\oli\AppData\Local\Temp\2163899450\sketch_merged.cpp
Sketch wird kompiliert ...
"C:\\Users\\oli\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_MKRWIFI1010 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8054 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR WiFi 1010\"" -DUSE_BQ24195L_PMIC "-IC:\\Users\\oli\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\oli\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\oli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.14\\cores\\arduino/api/deprecated" "-IC:\\Users\\oli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.14\\cores\\arduino/api/deprecated-avr-comp" "-IC:\\Users\\oli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.14\\cores\\arduino" "-IC:\\Users\\oli\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.14\\variants\\mkrwifi1010" "C:\\Users\\oli\\git\\ArduinoMakerWorkshopMinimal\\build\\sketch\\ArduinoMakerWorkshopMinimal.ino.cpp" -o "C:\\Users\\oli\\git\\ArduinoMakerWorkshopMinimal\\build\\sketch\\ArduinoMakerWorkshopMinimal.ino.cpp.o"
Bibliotheken werden kompiliert ...
Kern wird kompiliert ...
"C:\\Users\\oli\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_SAMD_MKRWIFI1010 -DARDUINO_ARCH_SAMD -DUSE_ARDUINO_MKR_PIN_LAYOUT -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x8054 -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino MKR WiFi 1010\"" -DUSE_BQ24195L_PMIC "-IC:\\Users\\oli\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\oli\\AppD [... cut ...]
"C:\\Users\\oli\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-size" -A "C:\\Users\\oli\\git\\ArduinoMakerWorkshopMinimal\\build/ArduinoMakerWorkshopMinimal.ino.elf"
Der Sketch verwendet 12244 Bytes (4%) des Programmspeicherplatzes. Das Maximum sind 262144 Bytes.
Globale Variablen verwenden 2988 Bytes (9%) des dynamischen Speichers, 29780 Bytes für lokale Variablen verbleiben. Das Maximum sind 32768 Bytes.

Verwendete Plattform Version Pfad
arduino:samd         1.8.14  C:\Users\oli\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.14
Command executed successfully.
Compilation completed successfully.

thelastoutpostworkshop, 2025-01-08:
I tried it with no problems, but from you screenshots the problems appears to come from another extension, Clang maybe ?

0x3b29, 2025-01-09:
I started removing all the extentions that could have an impact but the issue remained at first. It was somehow always tied to the Intellisense configuration. The config file had already lots of outdated fields in it so I decided to create a new profile in VS Code to start from scratch and it resolved the issue. Now the errors are gone and Intellisense finally works 👍

closed 2025-01-09 by 0x3b29