thelastoutpostworkshop/arduino-maker-workshop #23: Add caching to avoid calling the arduino-cli when it's not necessary

thelastoutpostworkshop, 2025-01-13:
For some arduino-cli commands, it does not make sense to call the arduino-cli every time.
A cache is implemented to solve this,

thelastoutpostworkshop, 2025-01-13:
@rudeb0t I have added the cache fonctionnality to the main branch, if you could test it on mac OS before, it would be great.

shawnpinciara, 2025-01-15:
Hey I've been using this extension for the past days for work... I've noticed as well that there are some thing that are slow: like choosing a board on the configuration tab (that i guess can be chached?)  or choosing the com port (can this be done faster? Arduino IDE is crazy fast with that), overall the experience is really nice and finally there is a good extension that doesn't need crazy configurations like platformio, so thank you so much for that :)

thelastoutpostworkshop, 2025-01-17:
@rudeb0t I am intending to change the way I do caching, I am planning to use the cliArg class instead of the cli class.  I want to cache everything but have time to live more specific for each type of cli commands

thelastoutpostworkshop, 2025-01-18:
@rudeb0t finally I decided not to make the change but added time to live parameter.
If you could test it, that would be nice.
Thanks.

closed 2025-01-24 by thelastoutpostworkshop