thelastoutpostworkshop/arduino-maker-workshop #24: libraires in the 'libraries' subfolder of the sketch's folder not found in compile

roblatour, 2025-01-13:
**Extension version** 0.6.2

**OS & version** W11

**Describe the bug** libraires in the 'libraries' subfolder of the sketch's folder not found in compile

**To Reproduce**
Steps to reproduce the behavior:

compile a sketch where the library to be uses is found only in the 'libraries' subfolder of the sketch's folder

**Expected behavior**

** Frist thanks very much for making this extension available, I've been using it for a couple weeks and it is very helpful. **

As I understand it, the Arduino IDE follows a specific hierarchy when looking for libraries during compilation. 

According to  https://arduino.github.io/arduino-cli/1.1/sketch-build-process/#location-priority here's the pecking order (from first to last)

1. The library is under a custom libraries path specified via the [--libraries option](https://arduino.github.io/arduino-cli/1.1/commands/arduino-cli_compile/#options) of arduino-cli compile (in decreasing order of priority when multiple custom paths are defined)

2. The library is under the libraries subfolder of the IDE's sketchbook or Arduino CLI's user directory

3. The library is bundled with the board platform/core ([{runtime.platform.path}/libraries](https://arduino.github.io/arduino-cli/1.1/platform-specification/#global-predefined-properties))

4. The library is bundled with the [referenced](https://arduino.github.io/arduino-cli/1.1/platform-specification/#referencing-another-core-variant-or-tool) board platform/core

The library is bundled with the Arduino IDE (this location is determined by the Arduino CLI configuration setting directories.builtin.libraries)

However, 0.6.2 of the extension appears to ignore libraries in the sketch folder under the "libraries" subfolder (point 2. above), and appears to look for them under the C:\Users\>user name>\OneDrive\Desktop\Documents\Arduino\libraries instead (at least on my machine).

Also, in 0.6.2 of the extension, I can't find where to set the sketchbook location ( as can be done in the Arduino IDE Files - Preferences) - I assume this is the workspace location opened in visual studio code (in which the .ino file is found).

**Screenshots**
n/a

**Outputs**
n/a

**Additional context**
Also, I don't

per1234, 2025-01-14:
> 2. The library is under the libraries subfolder of the IDE's sketchbook or Arduino CLI's user directory
>
>[...]
>
> the extension appears to ignore libraries in the sketch folder under the "libraries" subfolder (point 2. above),

The _sketch folder_ and the _sketchbook folder_ are two completely different things. So it is normal and expected that the libraries under the `libraries` subfolder of the _sketch folder_ will not be recognized.

> and appears to look for them under the `C:\Users\<user name>\OneDrive\Desktop\Documents\Arduino\libraries` instead

That is because the sketchbook folder location is configured to `C:\Users\<user name>\OneDrive\Desktop\Documents\Arduino` (this is the default).

>  I can't find where to set the sketchbook location

This is already tracked at the previous issue: https://github.com/thelastoutpostworkshop/arduino-maker-workshop/issues/10

Instructions are provided in that issue for configuring the sketchbook location (`directories.user`) by manually editing the [**Arduino CLI** configuration file](https://arduino.github.io/arduino-cli/latest/configuration/).

thelastoutpostworkshop, 2025-01-14:
Yes your are right : [sketchbook](https://arduino.github.io/arduino-cli/1.1/sketch-specification/#sketchbook)

roblatour, 2025-01-14:
yes.  The compile works using the Arduino IDE with the libraries only under the sketchbook folder.   Will wait for issue 10 to be resolved to see if then the code will compile using the AMW extension in the same way.

closed 2025-01-14 by thelastoutpostworkshop