thelastoutpostworkshop/arduino-maker-workshop #28: CLI : Failed to get Arduino CLI version information

sbonaime, 2025-01-24:
With new installed arduino-maker-workshop Version 0.4.4 in my Intel Macos Sequoi 15.2 I have the 
"Version 0.4.4 windows Arduino CLI on Macos" error 

[image]

I think the problem is that the wrong Arduino CLI is installed. In 
~/.vscode/extensions/thelastoutpostworkshop.arduino-maker-workshop-0.4.4/arduino_cli/
I only have a  win32 directory !

For some reasons arduino-maker-workshop does not installed the macos CLI. 
Any idea ? 

Thanks

sbonaime, 2025-01-24:
Macos Intel is not (yet) supported ! 
[#2 ](https://github.com/thelastoutpostworkshop/arduino-maker-workshop/discussions/2)

rudeb0t, 2025-01-24:
> Macos Intel is not (yet) supported ! [#2 ](https://github.com/thelastoutpostworkshop/arduino-maker-workshop/discussions/2)

Just added the x64 CLI binary in https://github.com/thelastoutpostworkshop/arduino-maker-workshop/pull/29

closed 2025-01-24 by thelastoutpostworkshop