thelastoutpostworkshop/arduino-maker-workshop #30: Arduino Maker Workshop resets the com port after programming a teensy 4.1

codewitch-honey-crisis, 2025-01-25:
**Extension version**
0.6.7

**OS & version**
Windows 11

**Describe the bug**
Every time you program a teensy device it causes the USB port to turn off and on. Arduino Maker Workshop responds to a port being disabled -even temporarily - by resetting the upload and monitor ports, meaning (a) you can't upload without setting the port each time and (b) you can't monitor without changing the monitor port after upload. It's maddening.

What Arduino Maker Workshop needs to do in order to work with devices like the Teensy, or the ESP32S3 in USB Serial JTAG mode, is to NOT reset the com port selection when a COM port goes offline. 

If you really insist that you MUST do this in order to make it droolproof then at the very least reset the com ports to the last selected ports.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a teensy 4.1, or an ESP32S3 devkit (using USB native serial jtag, not a dedicated UART chip), with more than one com port in your system, preferably with one you aren't using that will appear in the port list before the one you *are* using.
2. Select the board, and the upload and monitor ports
3. Upload a sketch
4. See that the port selection has changed even though you didn't want it to.

**Expected behavior**
I expect AMW to remember my port selection even when the port gets temporarily disconnected.

**Screenshots**
n/a

**Outputs**
n/a

**Additional context**
Thanks for looking into this.

thelastoutpostworkshop, 2025-01-30:
in step 4 the port selection change occurring, it's for the upload port ?

codewitch-honey-crisis, 2025-01-30:
Yes, both upload and monitor ports reset to their initial values

EFlynn7778, 2025-02-21:
Same issue with ESP32.  Very annoying and cumbersome.

It also does not honor the serial monitor auto-enable switch and it will turn on the serial monitor regardless.

Among other issues.....

thelastoutpostworkshop, 2025-02-23:
Added user setting to disable automatic port detection on Windows (v0.7.0).

closed 2025-02-23 by thelastoutpostworkshop