thelastoutpostworkshop/arduino-maker-workshop #31: Successful compile writes incorrect c_cpp_properties.json

MatAtBread, 2025-01-29:
**Extension version**
0.6.7

**OS & version**
Windows 11

**Describe the bug**
Intellisense offers to add a missing include (see below). I add it, and the squiggles disappear. I compile the file successfully. The just added include file is removed by the extension, and the squiggles re-appear.

** Additional information **
The added file (by intellisense) has the path:
```
"C:/Users/matthew/AppData/Local/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-cfea4f7c-v1/esp32c6/include/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos"
```
...which _shouldn't_ be necessary as it is under the already existing path (created by Extension/Arduino SDK) in the array:
```
"C:\\Users\\matthew\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.3-cfea4f7c-v1\\esp32c6\\include/**",
```
....however, Intellisense fails to locate it (path too long?) so it offers to add the freertos path.

It is possible this is a windows path separator `\` vs *nix path separator `/` issue??

**To Reproduce**
1. Accept the intellisense suggestion (the c_cpp_properties.json is updated)
2. Compile
3. The added path is missing from the c_cpp_properties.json

**Expected behavior**
The added path, even though it's a sub-path of an existing one, should be left in place.

MatAtBread, 2025-01-30:
Quick update: I tried installing and compiling everything in a WSL instance on the same Windows 11 installation. It compiles and updates the c_cpp_properties just fine. However it's a PITA to use as the COM ports have to be mapped through a TCP serial port server, which just adds a layer of pain :(

I think this is def a WIndows/Linux or path length issue.

thelastoutpostworkshop, 2025-01-30:
Thanks for conducting more tests, in WSL, you are on a linux distro, like Ubuntu right ?
So the problem is only on Windows 11 ?

MatAtBread, 2025-01-30:
Yep. The default Ubuntu distro.

MatAtBread, 2025-01-30:
Which I suppose means this could be a VSCode issue, since it's incorrectly not finding the required includes under subdued l Windows, but it does under WSL

thelastoutpostworkshop, 2025-01-30:
In order for me to test it on my side on Windows 11, can you provide a simple sketch that reproduce the problem and specifying the board you have selected.

MatAtBread, 2025-01-30:
I can try. Probably tomorrow! 

The board is a Seeed XIAO ESP32C6.

The sketch is pretty basic. It only uses the std libraries (WiFi, pins).

MatAtBread, 2025-01-30:
Ok, I was trying to create you a simple sketch, but I get a different set of errors. I _think_ this might be an issue with the order of the paths, My new sketch finds the WiFi.h header, but doesn't compile. The WiFi.h it finds is in a different location:

`C:\Users\matthew\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32c6\include\esp_wifi\include\esp_private\wifi.h`

...as opposed to my main project:

`C:\Users\matthew\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.1\libraries\WiFi\src\WiFi.h`

Both projects are configured to use the same board (XIAO_ESP32C6). I'm not familiar enough with the Arduino library/tools to know why two projects created to use the same boards/options should produce different C compiler includes. The relevant sections are:

* New project (won't compile as it finds the wrong WiFi.h)
```
      "includePath": [
        "C:\\Users\\matthew\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.1\\cores\\esp32/**",
        "C:\\Users\\matthew\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.1\\variants\\XIAO_ESP32C6/**",
        "C:\\Users\\matthew\\Documents\\Arduino\\libraries/**",
        "C:\\Users\\matthew\\git\\esp-playtime\\test-c6-sketch/**",
        "C:\\Users\\matthew\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.3-cfea4f7c-v1\\esp32c6\\include/**"
      ],
```
and the other main project which does compile, but has the OP I reported:
```
      "includePath": [
        "C:\\Users\\matthew\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.1\\cores\\esp32/**",
        "C:\\Users\\matthew\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.1\\variants\\XIAO_ESP32C6/**",
        "C:\\Users\\matthew\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.1\\libraries\\WiFi\\src/**",
        "C:\\Users\\matthew\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.1.1\\libraries\\Network\\src/**",
        "C:\\Users\\matthew\\Documents\\Arduino\\libraries\\ESP_Telnet\\src/**",
        "C:\\Users\\matthew\\Documents\\Arduino\\libraries/**",
        "C:\\Users\\matthew\\git\\esp-playtime\\maker-1/**",
        "C:\\Users\\matthew\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.3-cfea4f7c-v1\\esp32c6\\include/**"
      ],

```

MatAtBread, 2025-01-30:
Test sketch ("new") above:

```
#include <Arduino.h>
#include <Wifi.h>

void setup() {
    WiFi.mode(WIFI_STA);
}
void loop() {}
```

closed 2025-01-30 by MatAtBread

thelastoutpostworkshop, 2025-01-30:
You closed the issue ?

MatAtBread, 2025-01-30:
Erm no, not intentionally(!)

Although after I created the new (test) project, I couldn't re-create the issue. Maybe creating the new project and therefore running the arduino CLI installation fixed something?