thelastoutpostworkshop/arduino-maker-workshop #32: Stuck at board selection

araujosemacento, 2025-02-01:
Hey guys, I'm sorry about revisiting a issue that was deemed as stale, but I think I'm facing a similar situation to issue #3 as I'll be providing a gif to the recreation of the bug.

### **Extension version**
Latest version available in vscode's extensions marketplace (e.g. 0.6.7)

### **OS & version**
Windows 11 23H2 (OS comp 22631.4830)

### **Describe the bug**
I can successfully install platforms through the `Boards Manager`, but regardless of the option I attempt to select, the `Board Selection` screen stays stuck without a selected option on `Select a board below` dialog.

### **To Reproduce**
Steps to reproduce the behavior:
1. Open `Marker Workshop Home`
2. Install a board from `Boards Manager`
3. Open `Board Selection` page
4. Select a board among the avalaible
5. No updates are actually made

[image]

### **Expected behavior**
As shown in the [video guide](https://www.youtube.com/watch?v=rduTUUVkzqM) the selected board would be smoothly updated to the workspace, including Maker Workshop and arduino.json.

### **Additional Context**
Here I'll try and add some screenshots of Maker Workshop tab I hope may be useful to clarify the enviroment setup

[image]

I don't know if it is unrelated or not but the `Home` page says sketch name and folder name do not match, given I hadn't set a sketch name yet... Manually that is, wondering if that would solve the issue and how would I **properly** do it.

[image]

araujosemacento, 2025-02-01:
## Welp, it turns out I'm stupid...

After all, the home page was, indeed, all the troubleshooting that I needed. I feel **so dumb** about this now, but turns out that if you do simply _rename_ your **sketch file' name** to be *the same* as your **folder's name** The project will be properly recognized as an Arduino one. That's probably what the user from issue #3 found out as well and got a bit too embarassed to come back and update the issue. So, my bad guys, I'm sorry about seeking help over my own ignorance inside of the tool enviroment.

#### Keep up the great work and I'm sorry once again, thanks for your time and patience if you read it thus far.

closed 2025-02-01 by araujosemacento

thelastoutpostworkshop, 2025-02-01:
No problem, I will make the message "Sketch Name does not match folder name" more clear.