thelastoutpostworkshop/arduino-maker-workshop #33: How to create a new sketch

wazig, 2025-02-01:
When I try to make another sketch , I do not know how...

I tried every step I could think of and this should be a simple task.....grrr

Do I need to copy the so called workspace ?

thelastoutpostworkshop, 2025-02-01:
There are several ways to that, here is one :
- In VS Code, open a folder where you want sketch to be stored (File > Open Folder)
- Open the Arduino Maker Workshop Home tab, it will ask you to create a new sketch, this will create a new sketch inside the folder you opened before.

wazig, 2025-02-02:
Thank you for the education !

This worked and helped me a lot.... Being a 30 year programmer with LabView I was spoiled to the ease of use with LabView.
Feel going back in time but now i'am able to upgrade my Renault Twizzy's battery.

Thanks again !

PS, you get 5 star from me...

thelastoutpostworkshop, 2025-02-02:
Thanks for the stars!

closed 2025-02-02 by thelastoutpostworkshop