thelastoutpostworkshop/arduino-maker-workshop #34: FileSystem

InnuendoPi, 2025-02-06:
What is the recommended process pack and upload /data (ESP32/8266)?
Thx for this great extension!

rudeb0t, 2025-02-06:
I was working on getting this feature working in the extension. However, I
am currently in the middle of transitioning between jobs. I will be able to
resume working on it once things settle down a bit for me.

InnuendoPi, 2025-02-09:
As a temporary solution, I've published my vscode extension VSCODE-ESPFS on marketplace. It is only a mod of the [original extension by kash4kev](https://github.com/kash4kev/vscode-esp8266fs) with littlefs support.

PaulChristopher, 2025-09-14:
See https://github.com/thelastoutpostworkshop/arduino-maker-workshop/issues/56#issuecomment-3289749533 for a simple, temporary solution to upload files using a batch skript.

thelastoutpostworkshop, 2025-11-29:
It's currently in a separate web application : https://thelastoutpostworkshop.github.io/microcontroller_devkit/espconnect/