thelastoutpostworkshop/arduino-maker-workshop #35: Can't add Espressif ESP32 Wrover Module

nikryden, 2025-02-08:
Hi 
First, I want to say that it's a really nice extension!

Have a problem when i try to add Espressif ESP32 Wrover Module. 
In the board manager the Espressif ES32 is installed

[image]

But in board selection I can find it but it won't select it just loading. What am I missing?

[image]

thelastoutpostworkshop, 2025-02-08:
Reproduce the problem and copy here the output of "Arduino CLI"

nikryden, 2025-02-08:
No caching, running command...
Running Arduino CLI...
c:\Users\ryden\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.6.7-win32-x64\arduino_cli\win32\arduino-cli.exe
version --json

Cache hit:coreupdate-index
No caching, running command...
Running Arduino CLI...
c:\Users\ryden\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.6.7-win32-x64\arduino_cli\win32\arduino-cli.exe
config dump --json

No caching, running command...
Running Arduino CLI...
c:\Users\ryden\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.6.7-win32-x64\arduino_cli\win32\arduino-cli.exe
config get directories.user

Cache hit:coreupdate-index
No caching, running command...
Running Arduino CLI...
c:\Users\ryden\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.6.7-win32-x64\arduino_cli\win32\arduino-cli.exe
outdated --json

No caching, running command...
Running Arduino CLI...
c:\Users\ryden\.vscode\extensions\thelastoutpostworkshop.arduino-maker-workshop-0.6.7-win32-x64\arduino_cli\win32\arduino-cli.exe
board list --json

Cache hit:boardsearch

thelastoutpostworkshop, 2025-02-08:
Just to confirm, just before selecting the board, clear the output "Arduino CLI", and select your board.
You should have commands similar to this :
```
No caching, running command...
Running Arduino CLI...
c:\Users\Charles\Desktop\Public Repo\vscode-arduino\arduino_cli\win32\arduino-cli.exe
board details --fqbn esp32:esp32:esp32wrover --json
```
You have the same behavior with the Pico boards ?

nikryden, 2025-02-09:
Thank you for all your help and quick response.

I managed to fix the issue. 
The solution was to remove the project and create it again, and now it's working perfectly.

Thank you once again!

closed 2025-02-09 by nikryden