thelastoutpostworkshop/arduino-maker-workshop #36: Problems with installation of zip library

omarbaf, 2025-02-08:
I am having an issue with installing libraries.  If I choose a library from the 'Not Installed' tab, the library is getting installed properly.  However, if I download the same library as a .zip file from github, and try to install it as a zip file, it is not a complete success.  Even though the library folder does actually get installed in the 'libraries' folder of Arduino, the 'Zip installed' tab does not show the library, and also I get error when trying to compile my project using that library.  Any idea what I may be doing wrong.  If you want to test it out, I was using the "Bonezegei_Printf" library.   Thanks

thelastoutpostworkshop, 2025-02-08:
Is it this library that you tried to installed as zip : https://github.com/bonezegei/Bonezegei_Printf ?

omarbaf, 2025-02-08:
Yes.  Additional thing I found out while playing further with this.

Although the 'zip installed' tab doesnt show the installed library after installation is complete, if another library is added, then the 'zip installed' tab shows the missing library!

omarbaf, 2025-02-08:
on an unrelated note, is there a way to delete a library that is in the 'Updatable' tab without having to update it first?  i only see the option of update in that tab...

thelastoutpostworkshop, 2025-02-08:
> on an unrelated note, is there a way to delete a library that is in the 'Updatable' tab without having to update it first? i only see the option of update in that tab...

No, that's a nice catch, I will create this as another issue to allow it

thelastoutpostworkshop, 2025-02-08:
> Yes. Additional thing I found out while playing further with this.
> 
> Although the 'zip installed' tab doesnt show the installed library after installation is complete, if another library is added, then the 'zip installed' tab shows the missing library!

Ok thanks for the additional information, I will do more tests on my side

omarbaf, 2025-02-08:
Thanks.  By the way, great extension and appreciate your efforts and keeping it up to date.

I am also enjoying lots of your youTube videos.  Although I have been software engineer for over 30 years, I have picked up a few good ideas from your videos.

thelastoutpostworkshop, 2025-02-08:
Could you give me another library for testing to add as zip ?

omarbaf, 2025-02-08:
I just grabbed a random one.   https://github.com/RobTillaart/A1301/releases/tag/0.2.3.  Does the same thing.

I think it behaves the same way with any library that you can download the 'latest' as zip file.

thelastoutpostworkshop, 2025-02-09:
I know why it is not in the "Zip Installed" tab, that's because Bonezegei_Print is in the official Arduino registry, so it will be added to "Install & up to date:" tab or the "updatable" tab if it's an older version.

Now I discovered a problem while investigation, when you add a zip, the cache is not invalidated, this why you don't see it anywhere in the library manager.

omarbaf, 2025-02-09:
I didn't see it in any of the tabs, but after installing another library, it ended up in the 'zip installed' tab and not in the installed and updated.

I am not sure where you meant for it to be added, but I am glad you were able to duplicate it.

thelastoutpostworkshop, 2025-02-09:
This is strange, I do not get the same behavior as you.
Can you post a screenshot of your "zip installed" section.
Also what version of the extension you have ?

omarbaf, 2025-02-09:
The version number is 0.6.7.

As I mentioned before, since I installed another library, something caused the 'zip installed' tab to show the library.  Previous to installing another library the screen was blank.  If I take a screen shot it will show the library.

omarbaf, 2025-02-09:
I can try deleting the library and adding it again and post you a screen shot.  Let me give that a try.

omarbaf, 2025-02-09:
[image]

thelastoutpostworkshop, 2025-02-09:
ok thanks, I will release the fix in a couple of days.

closed 2025-02-09 by thelastoutpostworkshop