thelastoutpostworkshop/arduino-maker-workshop #41: installled board not installed?

otaolafranc, 2025-02-17:
**Extension version**
0.6.9

**OS & version**
linux ubuntu

**Describe the bug**
I have installed Arduino AVR Boards using the boards manager. nevertheless when I go to board selection -> arduino AVR Boards by arduino ->arduino mega  or mega 2560 I get the following errors:
```
CLI: Error from get board configuration, you may have to installed the board using the board manager
```
and also:
```
Unable to load Project Configuration
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Boars manager'
2. Click on 'Not installed'
3. Scroll down to 'Arduino AVR Boards'
4. install it
5. go to 'board Selection'
6. select 'Arduino AVR Boards by arduino'
7. select arduino mega or mega 2560 on select board

thelastoutpostworkshop, 2025-02-17:
Please reproduce the error and paste the output "Arduino CLI" and "Arduino Extension"

Zeppelin500, 2025-03-14:
Hello, first, thank you for this Software!

I use Linux Mint and have the same issue.
It seems like all boards are installed, but when I select one, nothing happens. The first time, I got the error message above. But I can't reproduce that. Now, nothing happens. The progress bar at the top just keeps going.

What information do you need?

[image]

Zeppelin500, 2025-03-14:
OK, after creating a scetch and installing the C++ extension, it work :-)

thelastoutpostworkshop, 2025-03-14:
Great!

closed 2025-03-14 by thelastoutpostworkshop