thelastoutpostworkshop/arduino-maker-workshop #53: ESP32S3 include errors: Cannot open avr/io.h

PaulChristopher, 2025-06-09:
Thanks for the project, just bought you a coffe some days ago! Here is a small issue I have...

**Extension version**
V0.7.2, Visual Studio Code 1.100.3, Built-in CLI v1.2.2 (2025-04-22T13:51:00Z)

**OS & version**
Windows 10

**Describe the bug**

[image]

Arduino UNO works fine, ESP32S3 Dev Board gives me an IntelliSense(?) error: The include is underlined red, VSC complains that it cannot open source file "avr/io.h" (dependency of "C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32\Arduino.h").

Looking into the above mentioned Arduino.h, there is no avr/io.h" included at all, strangely. So why does VSC complain?

All in all and important: The project compiles fine without any errors. Upload works, code works on the board. Directly after the compilation, the errors are gone. However they come back again after some time when editing the code.

C/C++-Compiler options are:
IntelliSense:
${default}

Include path:
C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\cores\esp32/**
C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\variants\esp32s3/**
C:\Users\XXX\Documents\Arduino\libraries/**
C:\Users\XXX\Documents\Arduino VS Code\test4/**
C:\Users\XXX\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32s3\include/**

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new sketch using the arduino-maker-workshop-plugin
2. From esp32 by Espressiv System, choose "ESP32S3 Dev Module"
3. Open the ino file and add "#include <Arduino.h>"
4. You will immediately get the above mentioned errors.
5. Compile. This works however fine. Errors are gone after compilation.
6. Start to edit the ino file again and type e.g. "Serial" to trigger IntelliSense. The above error will come back and Arduino.h will be underlinded again. The error is a pain when using .h or .cpp in the project. Each file will have include errors. 

[image]

**Expected behavior**
The should be no errors. It is unclear, why VSC complains about a missing avr/io.h which is not even part of ESP's Arduino.h

**Screenshots**
See above

**Outputs**
Sorry, unclear to me. Please advise, which outputs you need. As mentioned above, there are no errors in the CLI when compiling the project.

thelastoutpostworkshop, 2025-06-09:
That's a good tip

thelastoutpostworkshop, 2025-06-09:
I was not able to reproduce the problem, I have no red errors.
Here is the intellisense file generated with your example.
Do you have a particular board configuration for the ESP32S3 Dev Module other than the defaults ?  I would try with the same configuration.

[c_cpp_properties.zip](https://github.com/user-attachments/files/20662181/c_cpp_properties.zip)

PaulChristopher, 2025-06-10:
Here are my configurations. I did no change the defaults. As said: It seems to me that there is a conflict with preinstalled libraries in "C:\\Users\\XXX\\Documents\\Arduino\\libraries/**" ?? Maybe some of them a poorly made and meant for Arduino Uno only? I don't understand why intellisense complains about not beeing able to include avr/io.h. That's somehow nonsense...

[c_cpp_properties.zip](https://github.com/user-attachments/files/20680220/c_cpp_properties.zip)

My workaround so far is: I remove "C:\\Users\\XXX\\Documents\\Arduino\\libraries/**" from the includePath and I put my libraries in the same folder as the ino script or in the so calles src folder, see [https://arduino.github.io/arduino-cli/1.2/sketch-specification/#src-subfolder](https://arduino.github.io/arduino-cli/1.2/sketch-specification/#src-subfolder)

You might close the issue if you want.

PaulChristopher, 2025-06-10:
By the way: The line C:\Users\XXX\Documents\Arduino\libraries/** in c_cpp_properties.json seems to come back magically after each compilation. Is there a way to prevent that?