thelastoutpostworkshop/arduino-maker-workshop #58: Compile using folders on an alternate hard drive like D Errors ESP32

VirtualSniper04, 2025-06-30:
**Extension version**
Built-in CLI v1.2.2 (2025-04-22T13:51:00Z)

**OS & version**
Win 1064bit

**Describe the bug**
Issue compiling ESP32 sketches on a drive other than C drive.

When I try to create a sketch for the esp32 on my D drive, it fails right away saying "Arduino' is not a recognized as an internal or external command while trying to create the partitions.csv

All paths have been included in the c_cpp_properties.json file to path to original C drive location as needed.

Can build fine off D:\ drive using the standard Arduino UNO board after adding the include paths.  It just doesn't seem to work for the ESP32, specifically ESP32 Wrover is what I was using.

**To Reproduce**
Steps to reproduce the behavior:
1. Select a folder off D:\ drive
2. Create the sketch "test_connect_esp32"
3. Setup board ESP32 Wrover Module from "esp32 by Espressif Systems"
4. Click on Board Config Options and Review
5. Save All
6. Compile

**Expected behavior**
When doing this same procedure on my default C:\ drive under my documents, the program compiles and builds all the required folders (core, libraries, sketch) and files with a Compilation completed Successfully.

**Screenshots**
n/a

**Outputs**
"Arduino CLI"
Compile project starting...
FQBN: esp32:esp32:esp32wrover:PartitionScheme=minimal,DebugLevel=debug
Using board 'esp32wrover' from platform in folder: C:\Users\Viper\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0
Using core 'esp32' from platform in folder: C:\Users\Viper\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0

cmd /c if exist "D:\\=Ardrino=\\Custom_Robot_Vehicle\\test_connect_esp32\\partitions.csv" COPY /y "D:\\=Ardrino=\\Custom_Robot_Vehicle\\test_connect_esp32\\partitions.csv" "D:\\=Ardrino=\\Custom_Robot_Vehicle\\test_connect_esp32\\build\\partitions.csv"
'Ardrino' is not recognized as an internal or external command,
operable program or batch file.
Error during build: exit status 1

Used platform Version Path
esp32:esp32   3.2.0   C:\Users\Viper\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0
Command failed with code 1.
Compilation failed

"Arduino Extension"
Wrote configuration to d:\=Ardrino=\Custom_Robot_Vehicle\test_connect_esp32\.vscode\arduino.json

**Additional context**
I have also tried to just copy the folder after it's built from my C drive to D drive and update all paths as needed.  This again still doesn't work and gives same error.

"c_cpp_properties.json"
{
  "configurations": [
    {
      "name": "Arduino",
      "includePath": [
        "C:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\cores\\esp32/**",
        "C:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\variants\\esp32/**",
        "C:\\Users\\Viper\\Documents\\Arduino\\libraries/**",
        "D:\\=Ardrino=\\Custom_Robot_Vehicle\\test_connect_esp32/**",
        "C:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.4-2f7dcd86-v1\\esp32\\include/**"
      ],
      "forcedInclude": [
        "C:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\cores\\esp32\\Arduino.h"
      ],
      "compilerPath": "C:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-x32\\2411/bin/xtensa-esp32-elf-g++",
      "compilerArgs": [
        "C:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp-x32\\2411/bin/xtensa-esp32-elf-g++",
        "-MMD",
        "-c",
        "@C:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.4-2f7dcd86-v1\\esp32/flags/cpp_flags",
        "-w",

Current Workaround:
Just create sketches on C drive in my docs and backup to D drive.
        "-Os",
        "-Werror=return-type",
        "-DF_CPU=240000000L",
        "-DARDUINO=10607",
        "-DARDUINO_ESP32_DEV",
        "-DARDUINO_ARCH_ESP32",
        "-DARDUINO_BOARD=\"ESP32_DEV\"",
        "-DARDUINO_VARIANT=\"esp32\"",
        "-DARDUINO_PARTITION_default",
        "-DARDUINO_HOST_OS=\"windows\"",
        "-DARDUINO_FQBN=\"esp32:esp32:esp32wrover:UploadSpeed=921600,FlashFreq=80,FlashMode=qio,PartitionScheme=default,DebugLevel=none,EraseFlash=none\"",
        "-DESP32=ESP32",
        "-DCORE_DEBUG_LEVEL=0",
        "-DBOARD_HAS_PSRAM",
        "-mfix-esp32-psram-cache-issue",
        "-mfix-esp32-psram-cache-strategy=memw",
        "-DARDUINO_USB_CDC_ON_BOOT=0",
        "@C:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.4-2f7dcd86-v1\\esp32/flags/defines",
        "-ID:\\=Ardrino=\\Custom_Robot_Vehicle\\test_connect_esp32",
        "-iprefix",
        "C:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.4-2f7dcd86-v1\\esp32/include/",
        "@C:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.4-2f7dcd86-v1\\esp32/flags/includes",
        "-IC:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.4-2f7dcd86-v1\\esp32/qio_qspi/include",
        "-IC:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\cores\\esp32",
        "-IC:\\Users\\Viper\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\variants\\esp32",
        "@D:\\=Ardrino=\\Custom_Robot_Vehicle\\test_connect_esp32\\build/build_opt.h",
        "@D:\\=Ardrino=\\Custom_Robot_Vehicle\\test_connect_esp32\\build/file_opts",
        "D:\\=Ardrino=\\Custom_Robot_Vehicle\\test_connect_esp32\\build\\sketch\\test_connect_esp32.ino.cpp",
        "-o",
        "D:\\=Ardrino=\\Custom_Robot_Vehicle\\test_connect_esp32\\build\\sketch\\test_connect_esp32.ino.cpp.o"
      ],
      "defines": [
        "F_CPU=240000000L",
        "ARDUINO=10607",
        "ARDUINO_ESP32_DEV",
        "ARDUINO_ARCH_ESP32",
        "ARDUINO_BOARD=\"ESP32_DEV\"",
        "ARDUINO_VARIANT=\"esp32\"",
        "ARDUINO_PARTITION_default",
        "ARDUINO_HOST_OS=\"windows\"",
        "ARDUINO_FQBN=\"esp32:esp32:esp32wrover:UploadSpeed=921600,FlashFreq=80,FlashMode=qio,PartitionScheme=default,DebugLevel=none,EraseFlash=none\"",
        "ESP32=ESP32",
        "CORE_DEBUG_LEVEL=0",
        "BOARD_HAS_PSRAM",
        "ARDUINO_USB_CDC_ON_BOOT=0",
        "ARDUINO_BOARD=\\\"ESP32_DEV\\\"\"",
        "ARDUINO_VARIANT=\\\"esp32\\\"\"",
        "ARDUINO_HOST_OS=\\\"windows\\\"\"",
        "ARDUINO_FQBN=\\\"esp32:esp32:esp32wrover:UploadSpeed=921600,FlashFreq=80,FlashMode=qio,PartitionScheme=default,DebugLevel=none,EraseFlash=none\\\"\"",
        "ARDUINO_CORE_BUILD",
        "USBCON"
      ],
      "cStandard": "c17",
      "cppStandard": "c++17"
    }
  ],
  "version": 4
}

thelastoutpostworkshop, 2025-06-30:
I just created a folder on my H drive manually naming it "test" with a test.ino sketch inside it.
No problem compiling with the ESP32 Dev Module.
On your step 2 "Create the sketch "test_connect_esp32" do you do it with the extension ?

thelastoutpostworkshop, 2025-06-30:
I was able to fix it.  If I zip a .vsix file do you know how to install it in vscode to test it ?

VirtualSniper04, 2025-06-30:
To install a VS Code extension from a .vsix file, open the Extensions view, click the "More Actions" button (three dots), and select "Install from VSIX...". Then, navigate to and select the desired .vsix file, and finally reload VS Code if prompted. ?  I can try it out

thelastoutpostworkshop, 2025-06-30:
Yes, here is the zip file containing the .vsix

thelastoutpostworkshop, 2025-06-30:
oh the file is too large, let me find another way

thelastoutpostworkshop, 2025-06-30:
here is the link to download : https://drive.google.com/file/d/1H8MIAu1M-jt5MPZXUsf2OTJlomJNYtq6/view?usp=sharing

thelastoutpostworkshop, 2025-07-01:
Yes indeed, was able to reproduce the behavior.
It's the arduino-cli that's causing the problem, here in the documentation it's stated what characters are permitted in a sketch and folder name : https://arduino.github.io/arduino-cli/dev/sketch-specification/#sketch-folders-and-files

closed 2025-07-05 by thelastoutpostworkshop