thelastoutpostworkshop/arduino-maker-workshop #60: Using under WSL2 arduino.json gets the programmer setting wrong

OGMAvision, 2025-07-17:
**Extension version**
Version number of the Arduino Maker Workshop extension

Built-in CLI v1.2.2 (2025-04-22T13:51:01Z)

**OS & version**
Windows 11 Home.
Working under WSL2 in order to speed up builds.

**Describe the bug**
Every time I open the AMW configurations GUI, the programmer setting is rewritten with the first character in upper case. This breaks the uploading of code to an ESP32 board. It may be possible that it happens with other boards.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the AMW's GUI to change configurations and make sure Estpool is selected as the programmer.
2. Close the AMW GUI
3. Inspect the code inside .vscode/arduino.json
4. Note that now "programmer": "Esptool",
5. Try compiling and uploading a sketch, and it will fail.
6. Edit manually the content of .vscode/arduino.json
7. Set "programmer": "esptool", <--- Lower case.
8. Try to compile and upload, and it should work now.

**Expected behavior**
The GUI should make sure the variable is set in lowercase.

thelastoutpostworkshop, 2025-07-17:
I was able to reproduce the bug. 
I will fix it.

thelastoutpostworkshop, 2025-07-18:
It's now fixed in version 0.7.5.