thelastoutpostworkshop/arduino-maker-workshop #65: Programmer 'arduinoota125' not found

peno64, 2025-09-13:
Today I updated the addon in visual code. It was several months ago that I did an update. 
Now when I try to upload my sketch via the programmer which is Arduino OTA (I work with an arduino mega 2560 board with the Optiboot 8 bootloader) I get the following error:
Programmer 'arduinoota125' not found
I can still upload my sketch manually via the following command :
"C:\Users\mail\AppData\Local\Arduino15\packages\Optiboot8/tools/arduinoOTA.exe" -address 192.168.1.125 -port 65280 -t 30 -username arduino -password password -sketch "c:\git\ChickenGuard\build/ChickenGuard.ino.bin" -upload /sketch -b
However I do not have an arduinoota125.exe on my system.
I see that in .vscode\arduino.json that
"programmer": "Arduino OTA (192.168.1.125)",
is changed to:
"programmer": "arduinoota125",
This is probably done by the upgrade.
Changing this to 
"programmer": "arduinoota",
also doesn't help
still he says Programmer 'arduinoota' not found
Note that before this update I could upload without a problem. It is because I still could and that I copied that command .
Even adding C:\Users\mail\AppData\Local\Arduino15\packages\Optiboot8\tools to my path does not help.

Can you please help to make this work again?

thelastoutpostworkshop, 2025-09-13:
you can also change the programmer name in the arduino.json : 
arduinoOTA125,
I think the issue is the extension convert the programme id to lowercase

peno64, 2025-09-13:
This is my arduino.json:

{
  "port": "Arduino at 192.168.1.125",
  "configuration": "",
  "output": "build",
  "board": "Optiboot8:avr:megatris",
  "programmer": "arduino ota (192.168.1.125)",
  "useProgrammer": true,
  "configurationRequired": false,
  "monitorPortSettings": {
    "port": "COM3",
    "baudRate": 9600,
    "lineEnding": "\r\n",
    "dataBits": 8,
    "parity": "none",
    "stopBits": "one"
  },
  "optimize_for_debug": false
}

And with this, trying to upload it gives in the output:

Upload starting...
Programmer 'arduino ota (192.168.1.125)' not found

You are right

Changing to
"programmer": "arduinoOTA125",

Solves the issue. The problem is that it changes again when I go into the Maker Workshop Home page

thelastoutpostworkshop, 2025-09-13:
I see the problem is in the package, the programmer id is all lowercase.
Please try it in the Arduino IDE 2.3 to see if you get the error

peno64, 2025-09-13:
With IDE 2.3.6 it uploads without a problem:

Sketch uses 60660 bytes (23%) of program storage space. Maximum is 261120 bytes.
Global variables use 5642 bytes (68%) of dynamic memory, leaving 2550 bytes for local variables. Maximum is 8192 bytes.
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
OK
Sketch uploaded successfully

peno64, 2025-09-13:
Ah I guess you mean megatris.

In fact that is a modification I did. I copied mega to megatris. You can just use mega. The reason why I had to copy this board is because modification made to programmers.txt (I think it was that) were for some reason not seen. As if it is somewhere cached after being read the first time and you modifications are made they are not seen anymore. So I made a copy of mega first with name bis and later with tris to have a 'new' board and then it picked up the changes again for this 'new' board.
Maybe this caching  issue is solved in the meantime but I still use that configuration.

peno64, 2025-09-13:
Note again that before I updated your addin, programmer was this in arduino.mega.json:
"programmer": "Arduino OTA (192.168.1.125)",
After the update if became 
"programmer": "arduino ota (192.168.1.125)",
But now aparently it must be 
"programmer": "arduinoOTA125",
as you have discovered and this works when I change it manually in the arduino.mega.json file but when I go to Maker Workshop Home it changes again to
"programmer": "arduino ota (192.168.1.125)",

So something must have changed and it was not at my side because yesterday when I was still working with the old addin everything worked fine. It was only after updating the addin that this issue came.

peno64, 2025-09-13:
Thank you very much

closed 2025-09-14 by thelastoutpostworkshop