thelastoutpostworkshop/arduino-maker-workshop #68: Boards Manager not Showing the Latest Version of a Board

OGMAvision, 2025-09-25:
**Extension version**
Version number of the Arduino Maker Workshop extension: 1.0.5

**OS & version**
Windows 11

**Describe the bug**
I have added a board to the Boards Manager using the following URL:

https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json

I have been using versions 5.1.0 and 5.2.0 of this board.

If I open the JSON file from the URL above, I can see that version 5.2.1 is now available and listed for installation.

However, the GUI continues to show 5.2.0 as the latest version.

I don't know if this is a cache problem or a bug that only shows certain major versions.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the URL above and download/install the board version 5.1.0. 
2. Try to update, and in my case, I can only see version 5.2.0, but I don't see 5.2.1 in the dropdown.

**Expected behavior**
I expected to see version 5.2.1, since it is the one listed for download inside the JSON URL.

```
      "platforms": [
        {
          "category": "Raspberry Pi Pico",
          "name": "Raspberry Pi Pico/RP2040/RP2350",
          "url": "https://github.com/earlephilhower/arduino-pico/releases/download/5.2.1/rp2040-5.2.1.zip",
          "version": "5.2.1",
          "architecture": "rp2040",
          "archiveFileName": "rp2040-5.2.1.zip",
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

OGMAvision, 2025-09-25:
Update:

I ran the following and after restarting the IDE, 5.2.1 is listed.

```
arduino-cli core update-index --additional-urls https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json
arduino-cli core install rp2040:rp2040@5.2.1 --additional-urls https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json
```

If this is a cache issue, we could benefit from a button to re-download or refresh the last available version.

thelastoutpostworkshop, 2025-09-25:
Yes indeed, this a cache issue

thelastoutpostworkshop, 2025-09-25:
The cache is 12 hours, I think I should drop it to something like 2-3 hours